egmccaul/CIT-CSE480 #40: Update Individual Album Page Design

egmccaul, 2017-02-12:


egmccaul, 2017-02-24:
Was able to output SQL results with a photo path location set.
Able to sort results by oldest photo entry first, or newest photo entry first.
Pop-up modal implemented that allows users to switch through the photos displayed.
Set modal carousel to begin on selected photo.
Add in button to download photo from server location.

egmccaul, 2017-02-24:
To Do:
- [ ] Share photo to social media
- [ ] General Stylizing
- [ ] Push to Git
- [ ] Update description
- [ ] Update Image Title

petekal1323, 2017-02-24:
That's amazing you should just push

egmccaul, 2017-03-04:
Reverted to a two page design due to issues with pagination. Uses a current camera id session variable to generate the content. If you navigate to this page without this variable being set it will appear to be a blank page with the photo title.

egmccaul, 2017-03-04:
Added SQL to allow user to update image description and image title.

egmccaul, 2017-03-04:
Pushed to GitHub.

Pushed to test server location.

closed 2017-04-02 by egmccaul